khoih-prog/FlashStorage_STM32 #1: EEPROM write works only in a fully erased Bluepill with Chinese clone micro CS32F103C8T6.

jishnu-janardhanan, 2021-01-28:
I have a blue pill with chinese CS32F103C8T6 in it. EEPROM write (commit and put) works only after I completely erase the chip using STM32CubeProgrammer. Once there is any data other than 0xFF in any location, it will no longer write to any location. Every sample programs failed, after the initial run in that chip. Everything is working as it should on my other bluepill with Genuine (I hope) STM32F103C8T6.

khoih-prog, 2021-01-28:
Thanks for using the library and that interesting info.

I actually have no fake `chinese CS32F103C8T6` to know what's wrong with it. But what can you expect from a **fake, rejected or low-quality products**? The issue can be anywhere: hardware/board design, fake chip, bad flash, etc. and I don't think anyone will spend time to investigate why.

closed 2021-01-28 by khoih-prog

khoih-prog, 2021-08-17:
Hi @jishnu-janardhanan 

I've just written and published the new [**FlashStorage_STM32F1** library](https://github.com/khoih-prog/https://github.com/khoih-prog/FlashStorage_STM32F1) to take care of the special genuine / non-genuine STM32F103, CH32F103, CS32F103 boards, with **32/64/128KB** flash.

The flash size will be recognized automatically according to the registered flash size. Please try and report any issue  in [**FlashStorage_STM32F1** Issues](https://github.com/khoih-prog/FlashStorage_STM32F1/issues)

Best Regards,

---

## Changelog

### Releases v1.0.1

1. Fix mistake in initial releases

### Initial Releases v1.0.0

1. Initial release to support STM32F1/F3 boards with / without integrated EEPROM
2. Tested using genuine STM32F103, or CH32F103/CS32F103 boards with 32KB, 64KB or 128KB Flash Size
3. Configurable Flash Sector to use
4. FlashStorage size is 1024 bytes (including `1-byte valid` and `4-byte signature`)