khoih-prog/FlashStorage_STM32 #2: error  i got

alikirkbas, 2021-05-19:
dear sir or mam 
 i got this error
C:\Users\ali\Documents\Arduino\libraries\FlashStorage_STM32\src/utility/stm32_eeprom.h:61:23: fatal error: stm32_def.h: No such file or directory
 #include "stm32_def.h"

khoih-prog, 2021-05-19:
You have to first install the STM32 core v2.0.0+

Check the [**Prerequisites**](https://github.com/khoih-prog/FlashStorage_STM32#prerequisites).

In Arduino IDE Board Manager, search for `STM32 MCU` and install core v2.0.0

[image]

This is where [**stm32_def.h**](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/stm32/stm32_def.h) is.

In your local drive

`/.arduino15/packages/STMicroelectronics/hardware/stm32/2.0.0/cores/arduino/stm32/stm32_def.h
`

closed 2021-05-19 by khoih-prog

alikirkbas, 2021-05-19:
thnak you very much i am extremely ignorant to stm32.

Khoi Hoang ***@***.***>, 20 May 2021 Per, 00:48 tarihinde
şunu yazdı:

> Closed #2 <https://github.com/khoih-prog/FlashStorage_STM32/issues/2>.