khoih-prog/FlashStorage_STM32 #3: Not working on STM32F103

bakwc, 2021-06-01:
Tried to use this lib with blue pill, values resets after every reboot.

bakwc, 2021-06-01:
Finally found working solution:
https://ioprog.com/2018/05/23/using-flash-memory-on-the-stm32f103/
https://github.com/fduignan/stm32f103c8t6/tree/42ee0b39429b55f1d57458927a1d01ca1eb979ec
Works with `writeSector` / `readSector`

Please check why their solution works and your library doesn't work.

bakwc, 2021-06-02:
Hardware is fine, I used another library and it worked without any issues. It's a software bug.
There is and official STM32F103 flash access code:
```
int  writeSector(uint32_t Address,void * values, uint16_t size)
{              
    uint16_t *AddressPtr;
    uint16_t *valuePtr;
    AddressPtr = (uint16_t *)Address;
    valuePtr=(uint16_t *)values;
    size = size / 2;  // incoming value is expressed in bytes, not 16 bit words
    while(size) {        
        // unlock the flash 
        // Key 1 : 0x45670123
        // Key 2 : 0xCDEF89AB
        FLASH->KEYR = 0x45670123;
        FLASH->KEYR = 0xCDEF89AB;
        FLASH->CR &= ~BIT1; // ensure PER is low
        FLASH->CR |= BIT0;  // set the PG bit        
        *(AddressPtr) = *(valuePtr);
        while(FLASH->SR & BIT0); // wait while busy
        if (FLASH->SR & BIT2)
            return -1; // flash not erased to begin with
        if (FLASH->SR & BIT4)
            return -2; // write protect error
        AddressPtr++;
        valuePtr++;
        size--;
    }    
    return 0;    
}
```
Which works absolutely fine on my board.
There is STM32F1 with 64K and STM32F1 with 128K (both official), does your library handle both?

My board has 64K memory, so I use one of last memory page to access it:

```
writeSector(0x8000000 + 1024 * 60, &content, sizeof(content));
```

LuaksCz, 2021-07-25:
Hello i have the same issue, but i believe i have the original chip, it has STM markings and was bought from reputable vendor, so is there a way to check if its an original?

bakwc, 2021-07-25:
Here is an example that worked for me:
https://gist.github.com/bakwc/e5e8c255e03eb78ea4321c2c6179bdbf
(raw read / write, no external library required, see `example` function at the end of gist)

khoih-prog, 2021-07-25:
Hi @bakwc and @LuaksCz 

Thanks for all of your valuable inputs which help me to rethink about spending future time to investigate the issue.
Just want to be sure that you select the **correct board (C6/C8/CB, etc)** (flash size 32K, 64K or 128K) from the IDE / PIO.

It'll certainly take some time to order / receive as I don't have every kind of boards in the world to test now.

LuaksCz, 2021-07-25:
i have 64k stm32F103C with HID bootloader 2.2, the proposed solution by @bakwc is working, so thanks.

khoih-prog, 2021-08-03:
Just tested on **64KB BluePill STM32F103C8**, and without any issue of data retaining as mentioned.

So, I think there is some **issue with your code / usage of library**. As always, it's better to start with the simplest example to understand what the library is providing, and use it accordingly.

The library is designed for certain purpose and must be used correctly according to its design.

I'm closing the issue now and only reopen if you can prove your issue is caused by a library bug.

Good Luck,

---

### EmulatedEEPROM on BLUEPILL_F103C8

```
Start EmulatedEEPROM on BLUEPILL_F103C8
FlashStorage_STM32 v1.1.0
EEPROM length: 1024
EEPROM is empty, writing WRITTEN_SIGNATURE and some example data:
-> 104 105 106 107 108 109 110 111 112 113 114 115 116 117 118 119
Done writing to emulated EEPROM. You can reset now to test
```

#### Power OFF then ON

```
Start EmulatedEEPROM on BLUEPILL_F103C8
FlashStorage_STM32 v1.1.0
EEPROM length: 1024
EEPROM has been written.Signature = 0xBEEFDEED
Here is the content of the next 16 bytes:
-> 104 105 106 107 108 109 110 111 112 113 114 115 116 117 118 119
Clearing WRITTEN_SIGNATURE for next try
Done clearing signature in emulated EEPROM. You can reset now
```

---

### StoreNameAndSurname on BLUEPILL_F103C8

```
Start StoreNameAndSurname on BLUEPILL_F103C8
FlashStorage_STM32 v1.1.0
EEPROM length: 1024
EEPROM is empty, writing WRITTEN_SIGNATURE and some example data:
Insert your name : Your_Name

Insert your surname : Your_Surname

<< Your name: Your_Name
. Your surname: Your_Surname
 >> have been saved. Thank you!
```

#### Power OFF then ON

```
Start StoreNameAndSurname on BLUEPILL_F103C8
FlashStorage_STM32 v1.1.0
EEPROM length: 1024
Hi Your_Name
 Your_Surname
, nice to see you again :-)
Clearing WRITTEN_SIGNATURE for next try
Done clearing signature in emulated EEPROM. You can reset now
```

closed 2021-08-03 by khoih-prog

khoih-prog, 2021-08-17:
Hi all,

I've just written and published the new [**FlashStorage_STM32F1** library](https://github.com/khoih-prog/https://github.com/khoih-prog/FlashStorage_STM32F1) to take care of the special genuine / non-genuine STM32F103, CH32F103, CS32F103 boards, with **32/64/128KB** flash.

The flash size will be recognized automatically according to the registered flash size. Please try and report any issue  in [**FlashStorage_STM32F1** Issues)](https://github.com/khoih-prog/FlashStorage_STM32F1/issues)

Thanks for your research, reports and encouragement,

Best Regards,

---

## Changelog

### Releases v1.0.1

1. Fix mistake in initial releases

### Initial Releases v1.0.0

1. Initial release to support STM32F1/F3 boards with / without integrated EEPROM
2. Tested using genuine STM32F103, or CH32F103/CS32F103 boards with 32KB, 64KB or 128KB Flash Size
3. Configurable Flash Sector to use
4. FlashStorage size is 1024 bytes (including `1-byte valid` and `4-byte signature`)