khoih-prog/FlashStorage_STM32 #4: library not working on STM32F1

LuaksCz, 2021-07-23:
Hi, i am trying to use this library on STM32F1, i found this library through this site https://forum.arduino.cc/t/flashstorage-stm32-library-to-use-emulated-eeprom-now-supports-new-stm32-core-v2-0-0/693118 . There is shown STM32F1 as one of supported boards. However if i try to compile it it says that STM32F1 is not supported, so i would like to ask if i am reading it wrong or i am missing something? Also its giving me error for missing library stm32_def.h, dunno if its related.  

If its just wrong library, do you know which is the right one?

khoih-prog, 2021-07-23:
Hi @LuaksCz 

You're using the correct library which is supporting **STM32F1** as well as many other STM32. Check [**Currently supported Boards**](https://github.com/khoih-prog/FlashStorage_STM32#currently-supported-boards)

[image]

Be sure that you're using [**Arduino_Core_STM32 core**](https://github.com/stm32duino/Arduino_Core_STM32), not the [**rogerclarkmelbourne** core](https://github.com/rogerclarkmelbourne/Arduino_STM32) which is not supported.

If you're using the correct core, I'm afraid you're doing something wrong in the settings. Start with the simple example in the library, then compile and see, then moving forward gradually.

Good Luck,

closed 2021-07-23 by khoih-prog

khoih-prog, 2021-08-17:
Hi all,

I've just written and published the new [**FlashStorage_STM32F1** library](https://github.com/khoih-prog/https://github.com/khoih-prog/FlashStorage_STM32F1) to take care of the special genuine / non-genuine STM32F103, CH32F103, CS32F103 boards, with **32/64/128KB** flash.

The flash size will be recognized automatically according to the registered flash size. Please try and report any issue  in [**FlashStorage_STM32F1** Issues)](https://github.com/khoih-prog/FlashStorage_STM32F1/issues)

Thanks for your research, reports and encouragement,

Best Regards,

---

## Changelog

### Releases v1.0.1

1. Fix mistake in initial releases

### Initial Releases v1.0.0

1. Initial release to support STM32F1/F3 boards with / without integrated EEPROM
2. Tested using genuine STM32F103, or CH32F103/CS32F103 boards with 32KB, 64KB or 128KB Flash Size
3. Configurable Flash Sector to use
4. FlashStorage size is 1024 bytes (including `1-byte valid` and `4-byte signature`)