khoih-prog/FlashStorage_STM32 #5: EEPROM.commit() results in Hardware fault

KMeldgaard, 2021-08-31:
### Describe the bug
Dear Khoih-prog

I've been trying to use this library to write some configuration data to my MCU.
Everytime I execute the `EEPROM.commit()` function my MCU goes into hard fault - very rarely it is caught in some infinite loop.
I've traced the issue with the debugger to `HAL_FLASH_PROGRAM` which returns a `HAL_ERROR` (if it doens't just go into hard fault blink mode.

### Information
* MCU : STM32WB55CE
* Board: custom
* Core: Arduino_core_stm32 2.0.0 (https://github.com/stm32duino/Arduino_Core_STM32/tree/2.0.0)
* Build system: PlatformIO - core 5.1.1 - home 3.3.4
* host OS: Windows

#### aditional
* I'm using the `stm32duino/STM32duino FreeRTOS@^10.2.1` library to run "classic" FreeRTOS on my MCU.

### Steps to Reproduce
I have wrapped your library into a helper function which i run en `setup()` for testing purposes:
```
void use_khoih_prog() {
        for (int i = 0; i < 255; i++) {
            EEPROM.write(i, (uint8_t)i);
        }
        EEPROM.commit();
    }
```

Furthermore I've defined following:
```
// this defined is needed to use the FlashStorage_Stm32 package
#define STM32WB

// The developer allocated EEPROM page and start address
// FlashStorage_Stm32 uses 1 page for EEPROM
// ==> FLASH_END = 0x08055FFF
#define FLASH_PAGE_NUMBER 0x55
#define FLASH_BASE_ADDRESS 0x08055000
```
As I'm using the FULL_BLE_FW_STACK for the STM32WB55CE which starts on 0x08056000, And I want to make sure I'm not overwriting this (though that memory should be write protected).

I have also modified the linker-script MEMORY-section:
```
/* Specify the memory areas */
MEMORY
{
FLASH (rx)                 : ORIGIN = 0x08000000, LENGTH = 0x55000  /* =348'160B - BLE stack starts @ 0x08056000 on STM32WB55CE */
RETAIN_FLASH (rwx)         : ORIGIN = 0x08055000, LENGTH = 0x1000   /* =4096B - reserved for retained variables */
RAM1 (xrw)                 : ORIGIN = 0x20000004, LENGTH = 0x2FFFC
RAM_SHARED (xrw)           : ORIGIN = 0x20030000, LENGTH = 10K
}
```

### Expected behavior
0-->255 should be written to addresses 0x08055000 --> 0x080550FF

### Actual behavior

MCU goes into hardfault.

Hopefully your have an idea what could be the cause?

Regards
Kmeld

khoih-prog, 2021-08-31:
Hi @KMeldgaard 

Thanks for your interest in the library.

> I've traced the issue with the debugger to HAL_FLASH_PROGRAM which returns a HAL_ERROR (if it doens't just go into hard fault blink mode.

As you can see, with the new STM32WB, the issue comes either from the **bug of core** and/or the **mismatch between your `custom `board design / settings**. 

Just call the function `eeprom_buffer_flush()`, which is provided by the STM32 core, will create the crash.

I suggest that you create an [**MRE**](https://stackoverflow.com/help/minimal-reproducible-example), using `eeprom_buffer_flush()`, then post the issue on either 

1. [STM32 Forum, STM32Core](https://www.stm32duino.com/viewforum.php?f=35)
2. [Arduino_Core_STM32 issues](https://github.com/stm32duino/Arduino_Core_STM32/issues)

I'm afraid I can't help anything here as it's not the bug of the library and I currently don't have any STM32WB to test.

I'd appreciate if you can post some follow-ups here to benefit other users as well.

Good Luck,

closed 2021-08-31 by khoih-prog

khoih-prog, 2021-09-01:
You can also try this [**ArduinoCore cofre for STM32WB**](https://github.com/GrumpyOldPizza/ArduinoCore-stm32wb) and its [**EEPROM**](https://github.com/GrumpyOldPizza/ArduinoCore-stm32wb/tree/master/libraries/EEPROM) or [**SFLASH** Library](https://github.com/GrumpyOldPizza/ArduinoCore-stm32wb/tree/master/libraries/SFLASH) to see if it's any better.

Good Luck,