khoih-prog/FlashStorage_STM32 #7: STM32L432KBU6 – EEPROM will not retain after first cycle

izzeracing, 2022-11-28:
First off, great library! Thank you very much.

Works perfectly on a Nucleo L476RG, however...

Standalone STM32L432KBU6 with full chip erase, will write to EEPROM using EEPROM.update() and EERPOM.commit(), and retains perfectly after FIRST update/write with commit, but any update/write to the EEPROM after will not be retained after cycling power. The EEPROM is updated but it will not be retained after cycling power.

Arduino Core for STM32 2.3.0
Arduino 1.8.9

Let me know if there's anything else you need from me.

Many thanks,
Caleb

khoih-prog, 2022-11-28:
HI @izzeracing 

Thanks for using the library and your nice words.

> The EEPROM is updated but it will not be retained after cycling power.

I'm sorry that there are too many STM32 boards, and I don't have that `STM32L432KBU6` to know what's wrong with it.

I'm afraid only you can figure out why (core, board-related package / config files, flash overwritten by bootloader, design, etc.) because this is just Flash-storage emulated-EEPROM, not the real EEPROM.

If not solved, you can post the issue on STM32 or Arduino Forum to ask for help, certainly with [**MRE**](https://stackoverflow.com/help/minimal-reproducible-example) so that anyone can duplicate the issue.

I'd appreciate it if you can finally post the investigation results back here.

Good Luck,

izzeracing, 2022-11-28:
Hi @khoih-prog,

Thank you for the quick response. Understood – will do. I'm determined to figure it out and will follow up if I find a solution.

Best,
Caleb

closed 2022-11-29 by khoih-prog